adafruit/Adafruit_BNO08x #2: I2C not working on some boards

caternuson, 2020-10-28:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=19&t=171036
Guessing something with default I2C clock speeds?

Recreated with Feather nRF52840 Express and `more_reports` example:

[image]

[image]

ladyada, 2020-10-28:
i bet its the i2c buffer

@hathach can you increase i2c buffer on this device to 512 bytes or 1kbyte?

hathach, 2020-10-29:
> i bet its the i2c buffer
> 
> @hathach can you increase i2c buffer on this device to 512 bytes or 1kbyte?

hihi, unfortunately, I don't have the sensor board to test with, I will try my best to check  out the code and try to propose some modification for testing.

ladyada, 2020-10-29:
i know that we can end up reading a lot of bytes. for usart i had to increase buffer here
https://github.com/adafruit/ArduinoCore-samd/commit/fad2d91143d76e8c7e0eaa9c0d8f167230113e32

i2c buffer here is 256 bytes
https://github.com/adafruit/ArduinoCore-samd/blob/master/libraries/Wire/Wire.h#L74

hathach, 2020-10-29:
@caternuson can you quickly modify the `SERIAL_BUFFER_SIZE` to 256/512 and try it again. 
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/cores/nRF5/RingBuffer.h#L29

Since `SERIAL_BUFFER_SIZE` is used in multiple places e.g with uart tx/rx simply increased it here could cause quadruple the memory increased. I will make it more dynamic (with heap/malloc) so that uart could keep the current minimal 64 and I2C can have a larger one to work with :)

caternuson, 2020-10-29:
@hathach Thanks. Tried that. Same results with either 256 or 512 :(

Note that someone in the forum thread says they have same issue on STM32.

My clock suspicion is based on what I see in CircuitPython land:
https://github.com/adafruit/Adafruit_CircuitPython_BNO08x/blob/59fc2db68bba3851c184efbc6b272bef0e5cf027/examples/bno08x_simpletest.py#L15

And the results of these simple tests:
```python
Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit Feather nRF52840 Express with nRF52840
>>> import board
>>> i2c = board.I2C()
>>> i2c.try_lock()
True
>>> i2c.scan()
[]
>>>
```
```python
Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit Feather nRF52840 Express with nRF52840
>>> import board
>>> import busio
>>> i2c = busio.I2C(board.SCL, board.SDA, frequency=80000)
>>> i2c.try_lock()
True
>>> i2c.scan()
[74]
>>>
``` 

Datasheet claims it should be good up to 400kHz.

Maybe @siddacious can shed some light?

ladyada, 2020-10-29:
@caternuson what about 100khz or 200khz?

caternuson, 2020-10-29:
Hmmm. Well, those work.

**100khz**
```python
Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit Feather nRF52840 Express with nRF52840
>>> import board
>>> import busio
>>> i2c = busio.I2C(board.SCL, board.SDA, frequency=100000)
>>> i2c.try_lock()
True
>>> i2c.scan()
[74]
>>> 
```

**200khz**
```python
Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit Feather nRF52840 Express with nRF52840
>>> import board
>>> import busio
>>> i2c = busio.I2C(board.SCL, board.SDA, frequency=200000)
>>> i2c.try_lock()
True
>>> i2c.scan()
[74]
>>> 
```

But also, can't recreate original scan fail. It did that yesterday and I repeated it today for the REPL output above. But now:

**400khz** (5.3.1 doesn't have change to 100k default)
```python
Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit Feather nRF52840 Express with nRF52840
>>> import board
>>> i2c = board.I2C()
>>> i2c.try_lock()
True
>>> i2c.scan()
[74]
>>>  
```

and even:
```python
Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit Feather nRF52840 Express with nRF52840
>>> import board
>>> import busio
>>> i2c = busio.I2C(board.SCL, board.SDA, frequency=400000)
>>> i2c.try_lock()
True
>>> i2c.scan()
[74]
>>> 
```

ladyada, 2020-10-29:
kk thanks @siddacious can take a look later!

caternuson, 2020-10-29:
FWIW - it seems to work fine on Feather M4 Express.

[image]

[image]

csostena, 2020-10-29:
I also tried changing buffer size, with no luck.
My knowledge of the hardware, I2C and the SW environment is basic, so you are already far ahead with your experiments.
You may want to take a look at gammaburst's comment back at the Adafruit original forum (https://forums.adafruit.com/viewtopic.php?f=19&t=171036) where he talks about BNO's I2C timming issues.
Interesting that the STM32 also shows the problem, so it's not the nRF52 alone (lot's of issues on Nordic-DevZone forums about I2C timning and pullups).
Just got an ItsyBitsy Express BLE (oh no, another nRF52840)... will try with that and let you know.
Thank you and good luck.
CRS

csostena, 2020-10-29:
Just tried with "Adafruit ItsyBitsy nRF52840 Express - Bluetooth LE".
Same error (with all default settings).

garyrkey, 2020-10-30:
A couple more data points...
Feather HUZZAH ESP8266 WiFi using I2C does not work for me.
Feather M0 Adalogger works "out of the box" for rotation_vector example.
Both tried twice.

Thanks for working this issue!
Gary

ladyada, 2020-10-30:
hi folks please try the PR just posted! ^^

caternuson, 2020-10-30:
Works here.

`more_reports` example on Feather nRF52840 Express (same as original test and setup above):
[image]

garyrkey, 2020-10-30:
Works here, also.
Thanks!

closed 2020-10-30 by ladyada

csostena, 2020-10-30:
Working with 'Feather nRF52840 Express', 'ItsyBitsy nRF52840 Express BLE' and 'Nano 33 BLE (nRF52840)'.
Great work.
Thank you.