adafruit/Adafruit_BNO08x #5: Update RESET pin in examples for I2C scenario?

caternuson, 2021-01-28:
Lot's of people are probably going to use I2C via a STEMMA QT cable without connecting to reset. Ex:
https://forums.adafruit.com/viewtopic.php?f=22&t=174489

ladyada, 2021-01-28:
is it necessary?

tannewt, 2021-01-29:
What do you mean by "Update"? Remove?

caternuson, 2021-01-29:
Just need to update guidance somewhere. Guide? Code comment? etc. 

The guide doesn't show using RESET pin for I2C wiring:
https://learn.adafruit.com/adafruit-9-dof-orientation-imu-fusion-breakout-bno085/arduino#i2c-wiring-3072303-4
(looks like same for UART?)

If anyone uses example as is connected via a STEMMA cable, it won't work. Simply changing:
https://github.com/adafruit/Adafruit_BNO08x/blob/77ff9269e99da869b040739678de51ab07c179cf/examples/rotation_vector/rotation_vector.ino#L8
to -1 got it working.

Could also change the initialization:
https://github.com/adafruit/Adafruit_BNO08x/blob/77ff9269e99da869b040739678de51ab07c179cf/examples/rotation_vector/rotation_vector.ino#L10

ladyada, 2021-01-31:
oh yeah @caternuson ok please update the example to -1 then and submit a PR?

caternuson, 2021-02-02:
Fixed with #6

closed 2021-02-02 by caternuson