adafruit/Adafruit_BNO08x #7: setting output rate

jerabaul29, 2021-02-17:
I cannot find any indication of how to set the output rate. From the datasheet it looks like the output rate of most output streams of the chip are adjustable, with maximum output rates that depend on each stream type.

ladyada, 2021-02-17:
interval_us
https://github.com/adafruit/Adafruit_BNO08x/blob/master/src/Adafruit_BNO08x.h#L55

closed 2021-02-17 by ladyada

jerabaul29, 2021-02-17:
Ahh, thanks, sorry for missing that, nice :) Maybe showing this in the example / naming this in the online tutorial would be helpful ? :)

jerabaul29, 2021-02-17:
Mmmh, have you been able to test this and confirm it works? If I use something in this kind:

```cpp
#include "Arduino.h"

#include <Adafruit_BNO08x.h>

Adafruit_BNO08x bno08x{};
sh2_SensorValue_t sensorValue;

unsigned long micros_last;
unsigned long micros_crrt;

void setReports(void) {
  if (!bno08x.enableReport(SH2_ROTATION_VECTOR), 5000UL) {
    Serial.println("Could not enable rotation vector");
  }
}

void setup(void) {
  Serial.begin(115200);
  while (!Serial){
    delay(10);
  }

  Serial.println("------ booted ------");

  // initialize
  while (!bno08x.begin_I2C(0x4B, &Wire, 0)) {
    Serial.println("Failed to find BNO08x chip");
    delay(500);
  }

  Wire.setClock(400000);
  delay(50);
  
  setReports();

  micros_last = micros();

  Serial.println("----- Reading events -----");
  delay(100);
}

void loop() {
  Serial.println(F("start reading"));
  
  while(true){
    if (bno08x.wasReset()) {
      Serial.print("sensor was reset ");
      setReports();
      continue;
    }
  
    if (!bno08x.getSensorEvent(&sensorValue)) {
      continue;
    }
    else{
      switch (sensorValue.sensorId) {
      case SH2_ROTATION_VECTOR:
        micros_crrt = micros();
        Serial.print(F("micros: ")); Serial.print(micros_crrt); Serial.println(F(" | micros elapsed: ")); Serial.println(micros_crrt-micros_last);
        micros_last = micros_crrt;
        
        Serial.print("Rotation Vector - r: ");
        Serial.print(sensorValue.un.rotationVector.real);
        Serial.print(" i: ");
        Serial.print(sensorValue.un.rotationVector.i);
        Serial.print(" j: ");
        Serial.print(sensorValue.un.rotationVector.j);
        Serial.print(" k: ");
        Serial.println(sensorValue.un.rotationVector.k);
        break;
      }
    }
  }
}
```

I seems to always get a delta time of around 10,000 micro seconds ie 10 milliseconds ie 100Hz sampling rate.

caternuson, 2021-02-17:
```cpp
  if (!bno08x.enableReport(SH2_ROTATION_VECTOR), 5000UL) {
```
check parans? the int literal is part of the if conditional, not a parameter call to the func.

jerabaul29, 2021-02-18:
Oops, sorry, of course you are right, thanks :) .