adafruit/Adafruit_BNO08x #9: Possible reduced wiring requirements for SPI mode

kdharbert, 2021-03-01:
I have a BNO08x working with SPI and it requires 9 connections. I have a simple use case where the BNO08x is immediately initialized, runs constantly and is the only device on that SPI bus. Is there any chance I could simplify the wiring to use only 8 total connections?  

The driving use case is that I need to mount it to a Teensy 4.1 neatly.  I can easily make a dummy microSD adapter that can map the 8 microSD card adapter pins on the Teensy to the the BNO08x. You can then just insert the unit into the onboard microSD slot on the Teensy.  However, I need to salvage one pin from the BNO08x config to make it work.  

The microSD port has VCC,GND,SCK,MISO,MOSI and 3xDigital ports.  In SPI mode, the BNO08x requires CS,RST,INT and WAKPIN...which requires 4 digital pins and I only have three available.  If the simplistic use case allows for saving one wire on the BNO08x, I can make this work.  Any chance this is possible?

owennewo, 2021-03-28:
@kdharbert - you might be able to tie CS low as it is the only SPI device. I'd say there would be a 50% chance that would work (there is a chance that the BNO08x needs a CS toggle).  If this works then you'd save 1 pin.

Switching to I2C or UART-RVC interface would save two pins.

ladyada, 2021-05-29:
you can tie RST to the microcontroller RST but i really thing CS is required. I2C is another good idea!

closed 2021-05-29 by ladyada