adafruit/Adafruit_BNO08x #18: Possible solution to BNO08x not running on plain ESP32

norbinz, 2022-07-08:
I noticed this forum discussion describing trouble running a BNO08x on a plain ESP32:
https://forums.adafruit.com/viewtopic.php?f=19&t=191933

This message by adafruit2 reports no luck identifying the culprit:
https://forums.adafruit.com/viewtopic.php?f=19&t=191933#p930175

I reproduced the malfunction and created a possible fix:
https://forums.adafruit.com/viewtopic.php?f=19&t=191933&start=15#p931069

I hope that helps!

ladyada, 2022-07-10:
neat! :) can you try submit a PR for that fix to github? that's the best way for us to test it - and if it works, merge it into the library!

norbinz, 2022-07-11:
Hi Ladyada,
Sorry, no time or desire to learn git's way of communicating.

ladyada, 2022-07-11:
ok @caternuson can you repro?

caternuson, 2022-07-12:
At this point, just confirming recreating original issue using a QT Py ESP32 Pico with BNO08x breakout connected via STEMMA QT cable.

Shows up in a testbed I2C scan:
[image]

Running `more_reports` example sketch modified for QT Py ESP32 Pico's STEMMA QT Wire:
```cpp
  if (!bno08x.begin_I2C(BNO08x_I2CADDR_DEFAULT,&Wire1)) {
```
fails to run:
[image]

caternuson, 2022-07-12:
Issue persists even after updating `i2chal_open()` as suggested in forum thread (linked above).

[image]

Same behavior even after a power cycle of entire setup.

Note that this is failing in the call to `i2c_dev->begin()`:
https://github.com/adafruit/Adafruit_BNO08x/blob/4568d94c86bec8690c2d45de78b11849b157785d/src/Adafruit_BNO08x.cpp#L90

So it doesn't seem like `i2chal_open()` is even being called.

norbinz, 2022-07-13:
I added info to the forum thread to hopefully help duplicate the behavior:
https://forums.adafruit.com/viewtopic.php?f=19&t=191933&p=932302#p932302

caternuson, 2022-07-13:
Thanks for the additional information. Actually, my problem was me! I was forgetting about the need to also call `Wire1.setPins(SDA1, SCL1);` when dealing with the secondary port on the ESP32. Adding that extra bit to my sketch and it's working now. And it's working for me without the additional pull up resistor.

Your updates to `i2chal_open()` seem to be critical though.

With current library code:
[image]

With updated `i2chal_open()`:
[image]

Looks good! I'll PR these changes. **Thanks for figuring this out!**

norbinz, 2022-10-18:
This issue continues causing trouble for some Adafruit customers. For example:
https://forums.adafruit.com/viewtopic.php?p=944452#p944452

It's been several months, and I forgot that I had already debugged the problem, so I debugged it again!

caternuson, 2022-10-18:
Thanks for the reminder. Just submitted a pull request:
https://github.com/adafruit/Adafruit_BNO08x/pull/21

caternuson, 2022-11-01:
Closing. Should be fixed with #21.

closed 2022-11-01 by caternuson