adafruit/Adafruit_BNO08x #22: No way to get timestamp from sensor report?

joshgalvan, 2023-08-15:
Just the title.

Sparkfun's library with their BNO080 board has the ability to return the timestamp, and I know the timetamp is handled somewhere within this library but is never made available from the API it seems. Could this be added? It's a native feature of the sensor.

joshgalvan, 2023-08-15:
It seems the timestamp directly from the sensorValue variable will output a value from 500-10,000 randomly. Normally the values it outputs is around 4,294,900,000, then will randomly output 700 for some reason. In my use case this is only happening on Raw Accelerometer data.

closed 2023-09-05 by joshgalvan

YuriSizuku, 2024-02-02:
> It seems the timestamp directly from the sensorValue variable will output a value from 500-10,000 randomly. Normally the values it outputs is around 4,294,900,000, then will randomly output 700 for some reason. In my use case this is only happening on Raw Accelerometer data.

How do you solve this problem ?

joshgalvan, 2024-02-02:
@YuriSizuku Via the library I didn't. I just timestamped everything myself using `millis()`. Not as accurate technically but works well enough I suppose.

YuriSizuku, 2024-02-03:
> @YuriSizuku Via the library I didn't. I just timestamped everything myself using `millis()`. Not as accurate technically but works well enough I suppose.

Thank you for your replay. I wonder whether the result is by sequence or not. So that it has timestamp. It is very stronge that it often report a value nearly 4,294,900,000.