adafruit/Adafruit_BNO08x #23: Tare function for the BNO085 chips

TheJayOli, 2023-10-12:
Hello!

Is there currently any plans do implement the Tare function for the BNO085 chips to the library?

Is there currently a way to do this and I'm just not seeing it?

Thank you!

codydg, 2024-01-10:
I also would like to see tare support!

I see that `sh2.h` has the functionality [already included](https://github.com/adafruit/Adafruit_BNO08x/blob/master/src/sh2.h#L568-L590), so I'd expect it to be an easy addition, but I'm not sure how those static functions are associated with any particular instance of a `Adafruit_BNO08x`.

KeiranCantilina, 2024-02-15:
I too would like to see implementation of the tare functions.

alexanderwallbaum, 2024-05-23:
As @codydg pointed out already, it is pretty simple to implement. The tare function could look something like this:

Adafruit_BNO08x.cpp: 
```
...
/**
* Used to tare the sensor around all axis, using the gaming rotation vector as the basis
*/
bool Adafruit_BNO08x::tareSensor() {

  int status = sh2_setTareNow(SH2_TARE_X | SH2_TARE_Y | SH2_TARE_Z, SH2_TARE_BASIS_GAMING_ROTATION_VECTOR);
  
  if (status != SH2_OK) {
    return false;
  }

  return true;
}

/**
* Used to tare the sensor around the z-axis, using the gaming rotation vector as the basis
*/
bool Adafruit_BNO08x::tareSensorZ() {

  int status = sh2_setTareNow(SH2_TARE_Z, SH2_TARE_BASIS_GAMING_ROTATION_VECTOR);
  
  if (status != SH2_OK) {
    return false;
  }

  return true;
}

/**
* Used to persist the tare on the sensor. This is necessary to keep the tare after a reset of the sensor. 
* After the tare has been persisten it cannot be cleared / reset without a new tare
*/
bool Adafruit_BNO08x::persistTare() {
    
  int status = sh2_persistTare();

  if (status != SH2_OK) {
    return false;
  }

  return true;
}

/**
* Used to reset the tare on the sensor, if it has not been persisted.
*/
bool Adafruit_BNO08x::clearTare()
{
  int status = sh2_clearTare();

  if (status != SH2_OK) {
    return false;
  }

  return true;
}
...
```

<br />

Adafruit_BNO08x.h:

```
class Adafruit_BNO08x {
public:
    ...
    bool tareSensor();
    bool tareSensorZ();
    bool persistTare();
    bool clearTare();
    ...
}
```

shivenpatel399, 2024-12-02:
hey guys,

can you guys send me the github library path that have the tare function. The Adafruit BNO08x library I'm using seems to not have that function. More likely its outdated?