adafruit/Adafruit_BNO08x #32: Cannot create more than one instance

dhalbert, 2024-08-14:
Attempting to create more than one `Adafruit_BNO08x` fails, because `i2c_dev` is static, and gets discarded and re-created on the second call to the `Adafruit_BNO08x` constructor. I believe the `i2c_dev` should be a field in the device object.

See https://discord.com/channels/327254708534116352/537365760008257569/1273319381597687828 for a thread about this.

dhalbert, 2024-08-14:
I'm not sure if the `sensor_id` argument is necessary, since different instantiations could be distinguished by object identity.

dhalbert, 2024-08-14:
This is not so easy to fix, because the underlying library we use, https://github.com/ceva-dsp/sh2, is not re-entrant and only supports one sensor at a time. The SparkFun library is derived from ours; this problem was brought up there:
https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/issues/3
and then an issue was opened on the original library:
https://github.com/ceva-dsp/sh2/issues/9
This was November 2023, not sure if there will be any movement on this.

iLoveAndyBaker, 2024-08-21:
It's not just one sensor per bus either - it's just ONE sensor. Period.

VICLER, 2024-08-23:
I found out that the SparkFun BNO080 old library (https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library) works well with more than one BNO085 sensors on i2c bus. There are 2 addresses, so you can get up to 4x bno085 working on esp32 (with 2x i2c ports). But it does not work with SPI for some reason.. For my project I need 5x sensors, so I am using sparkfun bno080 for 4x sensors over i2c and one with sparkfun bno08x over spi.

Lunatunny, 2024-11-06:
[ras-marques](https://github.com/ras-marques) found that the BNO080 library from SlimeVR supports multiple BNO085 sensors just fine too. I grabbed their BNO080.cpp and .h (https://github.com/SlimeVR/SlimeVR-Tracker-ESP/tree/a4a9778f62d0b962c46898d03e9ce70171a1ca12/lib/bno080) and they worked perfectly on my ESP32 through the arduino IDE. For implementation examples, checkout https://github.com/SlimeVR/SlimeVR-Tracker-ESP/blob/a4a9778f62d0b962c46898d03e9ce70171a1ca12/src/sensors/bno080sensor.cpp#L33 and https://github.com/ras-marques/ImmersiveGloves/blob/main/Firmware/RightHand/usb1/usb1.ino

VICLER, 2024-11-06:
> [ras-marques](https://github.com/ras-marques) found that the BNO080 library from SlimeVR supports multiple BNO085 sensors just fine too. I grabbed their BNO080.cpp and .h [...]

That is the same SparkFun BNO080 library..

Lunatunny, 2025-01-11:
oh LOL that's literally what you said how did i not see that