adafruit/Adafruit_BNO08x #34: Need interrupt pin for I2C

aaronw2, 2024-10-06:
It seems that for reliable i2c support the interrupt pin should be checked before accessing it. This is what the SparkFun BNO085 library does. Failure to check this may result in i2c failures.

caternuson, 2024-10-07:
Please post the most simple Arduino sketch that demonstrates the issue.

brightproject, 2024-12-10:
Good day @caternuson and @aaronw2 🙂
I'm not entirely sure that I have a manifestation of the problem described above, but still.
Example code is this:

> https://github.com/adafruit/Adafruit_BNO08x/blob/master/examples/quaternion_yaw_pitch_roll/quaternion_yaw_pitch_roll.ino

I use `stm32f411` + `bno08x`, connection via `i2c`
If I flash the `mcu`, and I don't disconnect the device from the power supply, the example starts immediately and the `i2c` bus starts without failures:

> 4865    3       -4.36   -0.13   -0.42
> 4870    3       -4.35   -0.09   -0.40
> 4869    3       -4.34   -0.06   -0.38
> 5350    3       -4.32   -0.01   -0.36
> 4877    3       -4.30   0.04    -0.35
> 4880    3       -4.29   0.07    -0.35
> 5352    3       -4.27   0.11    -0.35

If I turn off the power to the device and power it on again, the `i2c` bus fails and does not find the address:

> Adafruit BNO08x test!
> I2C address not found
> Failed to find BNO08x chip

The only thing I added to the code:

> #define sq(x) ((x)*(x))

Because when assembling the firmware for `stm32`, the framework kernel did not have this define.
It is likely that this is a problem with the voltage on the `i2c` bus, the resistors are too big, or something else exists, but it is also possible that this is what the author of the issue writes about.

Maybe it is really possible to somehow rewrite the `i2c` bus startup function?

```
bool Adafruit_BNO08x::begin_I2C(uint8_t i2c_address, TwoWire *wire,
                                int32_t sensor_id) {
  if (i2c_dev) {
    delete i2c_dev; // remove old interface
  }

  i2c_dev = new Adafruit_I2CDevice(i2c_address, wire);

  if (!i2c_dev->begin()) {
    Serial.println(F("I2C address not found"));
    return false;
  }

  _HAL.open = i2chal_open;
  _HAL.close = i2chal_close;
  _HAL.read = i2chal_read;
  _HAL.write = i2chal_write;
  _HAL.getTimeUs = hal_getTimeUs;

  return _init(sensor_id);
}
```

caternuson, 2024-12-10:
@brightproject Do you see the same behavior with the rotation_vector example?
https://github.com/adafruit/Adafruit_BNO08x/blob/master/examples/rotation_vector/rotation_vector.ino
(hopefully can run that unmodified on stm32)

brightproject, 2024-12-11:
Got data on the `i2c` bus after the first power-up after power-off:

> BNO08x Read Example
> BNO08x found!
> Setting desired reports

> read to 0x7F nak
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF

After reset:

> BNO08x Read Example
> I2C address found
> Setting desired reports
> Rotation vector enabled
> Output in form roll, pitch, yaw
> Reading events
> BNO08x found!
> 
> write to 0x18 ack data: 0x00 0x00 0x0F
> 0xFF
> 0xFF
> 0xFA 0xC0 0x00 0x00 
> write to 0x40 ack
> write to 0x00 ack
> 
> write to 0x00 ack data: 0x00 0x00 0x00 0x00 
> write to 0x00 ack data: 0x40 
> write to 0x7E ack
> write to 0x00 ack data: 0x00 

I'm trying to solve the problem of no startup on the `i2c` bus with this hack:

```
// Try to initialize!
if (!bno08x.begin_I2C()) {
// delay(100);
//if (!bno08x.begin_UART(&Serial1)) { // Requires a device with > 300 byte UART buffer!
//if (!bno08x.begin_SPI(BNO08X_CS, BNO08X_INT)) {
Serial.println("Failed to find BNO08x chip");
while (1) {
delay(120);
NVIC_SystemReset();
}
}
```

Firstly, I increased the delay time

> delay(120);

Secondly, I added a forced reboot of the `stm32` in case of absence of data on the `i2c` bus

> NVIC_SystemReset();

After changing the code I get the following data on the `i2c` bus and in the serial port:

`Serial port`

> Adafruit BNO08x test!
> I2C address not found
> Failed to find BNO08x chip

`I2c bus`

> read to 0x55 nak
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> read to 0x55 nak
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF

`Serial port`

> Adafruit BNO08x test!
> I2C address not found
> Failed to find BNO08x chip
> Adafruit BNO08x test!

`I2c bus`

> read to 0x7F nak
> 0xFF
> 0xFF
> write to 0x44 ack data: 0x00 
> write to 0x00 ack

`Serial port`

> BNO08x Found!
> Setting desired reports
> Reading events
> sensor was reset
> Setting desired reports

`I2c bus`

> write to 0x00 ack data: 0x00 0x00 0x00 0x00 
> write to 0x00 ack data: 0x20 
> write to 0x00 ack data: 0x00 
> read to 0x7F nak
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> 0xFF
> write to 0x00 ack data: 0x00 
> write to 0x20 ack data: 0x00 0x01 
> write to 0x01 ack data: 0x00 0x00 0x00 0x00 0x00

And finally I get the data in `Serial port`:

> 469574  0       -35.80  -7.13   -6.79
> 5500    0       -35.82  -7.03   -6.75
> 2972    0       -35.81  -7.07   -6.78
> 3670    0       -35.81  -7.11   -6.80

Another report, without data from the `i2c` bus:

> Adafruit BNO08x test!
> I2C address not found
> Failed to find BNO08x chip
> Adafruit BNO08x test!
> I2C address not found
> Failed to find BNO08x chip
> Adafruit BNO08x test!
> BNO08x Found!
> Setting desired reports
> Reading events
> sensor was reset Setting desired reports
> 469035 0 -0.18 -0.15 -7.91
> 5103 0 -0.26 -0.08 -7.94
> 2952 0 -0.29 -0.06 -7.97
> 3475 0 -0.31 -0.04 -8.00
> 2950 0 -0.32 -0.00 -8.01
> 3904 0 -0.32 0.01 -8.03
> 5332 0 -0.31 0.01 -8.04

It is clear that the code makes two attempts to connect in a normal way, then resets the `stm32` and the third connection is successful and data starts to arrive.

I did not find the reason for such strange behavior of the `bno08x` chip, the wires from the mcu to the sensor are as short as possible, the resistors are `4.7 kOhm`, the voltage is `3.3 volts`.

Perhaps someone will have other solutions to the problem.